r-map/rmap #26: Lcd display display only time was not set text

dcast78, 2015-08-05:
Lcd display only show "R-map project" and "time was not set"

pat1, 2015-08-06:
strange ....
I need the debug messages printed on serial with compile options
Some more lines (for some mqtt publish cycles) will be usefull

dcast78, 2015-08-06:
Please remind: to resolve timeout un ntp response, we have disabled ntp sync on boot.
Thi is a minor bug with low priority
[image]

Terminal ready
#54,0,0/105,20,-,-/B12101
#payload:{"v":30377}
#B12101:30377
#{"v":30377}
#mqttrootpath:rmap/my_user/1137625,4449216/rmap/254,0,0/105,20,-,-/B12101
#payload:{"v":30377}
#Started; version: "1.0-470:474M" jsonrpc attuatore sensor ethernet radiorf24 lcd
#Configuration loaded
#sampletime:5
#mqttrootpath:rmap/my_user/1137625,4449216/rmap/
#mqttserver:rmap.cc
#mqttuser:my_user
#mqttpassword:rmy_password
#key:0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15, 
#iv:0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15, 
#ntpserver: pool.ntp.org
#thisnode: 0
# >> CONFIGURE NEW SENSOR: 0
#driver:I2C
#node:1
#type:TMP
#address:72
#mqttpath:254,0,0/105,200,-,-/
#NEW driver: I2CTMP
# >> CONFIGURE NEW SENSOR: 1
#driver:RF24
#node:1
#type:TMP
#address:72
#mqttpath:254,0,0/105,20,-,-/
#NEW driver: RF24TMP
#Try to configure Ethernet using DHCP
#My IP address is 192.168.1.133
#Netmask is 255.255.255.0
#Gateway IP address is 192.168.1.1
#DNS IP address is 192.168.1.1
#time was not set by the sync provider
#mqttid: 00:00:00:00:00:01
#error on mqtt connect
#00:00:08 01 01 1970
#mqtt disconnected
#mqtt reconnected
#mqtt subscribed to: rpc/master
#mqtt reconnected
#Repeats
#ethernet maintain: nothing happened
#The time has never been set
#NULL
#prepare: 0 ok.
#Radio Sending... prepare
#receive: {"jsonrpc":"2.0","result":{"waittime":500},"id":0}
#its a valid response
#waittime: 500
#prepare: 1 ok.
#wait for ms: 500
#getjson: 0
#looping over ajson object: 
#B12101:30633
#{"v":30633}
#mqttrootpath:rmap/my_user/1137625,4449216/rmap/254,0,0/105,200,-,-/B12101
#payload:{"v":30633}
#getjson: 1
#Radio Sending... getjson
#looping over ajson object: 
#B12101:30371
[... 218 lines cut ...]
#payload:{"v":30371}
#Repeats
#ethernet maintain: nothing happened
#The time has never been set
#NULL
#prepare: 0 ok.
#Radio Sending... prepare
#receive: {"jsonrpc":"2.0","result":{"waittime":500},"id":0}
#its a valid response
#waittime: 500
#prepare: 1 ok.
#wait for ms: 500
#getjson: 0
#looping over ajson object: 
#B12101:30671
#{"v":30671}
#mqttrootpath:rmap/my_user/1137625,4449216/rmap/254,0,0/105,200,-,-/B12101
#payload:{"v":30671}
#getjson: 1
#Radio Sending... getjson
#looping over ajson object: 
#B12101:30371
#{"v":30371}
#mqttrootpath:rmap/my_user/1137625,4449216/rmap/254,0,0/105,20,-,-/B12101
#payload:{"v":30371}

pat1, 2015-09-06:
need check with the last firmware

commit a84af164 (2015-10-09): close #26
closed 2015-10-09 by pat1