r-map/rmap #33: remote update firmware

pat1, 2015-09-28:
should be great the possibility to upgrade the firmware from a remote site.
brain storm on it:
a specific protocol to write firmware to a SD card
reboot
use this or somethings like this to unpgrade firmware from SD:
https://github.com/thseiler/embedded/tree/master/avr/2boots/

dcast78, 2016-04-28:
Brainstorming ideas...

For a remote upgrade it's useful to use two firmware images (eg: two files on sdcard).
- Tested firmware for secure boot
- New firmware for upgrade
If new firmware fail to boot (triggered by watchdog restart?) the bootloader automatically revert to old tested software. 
When the new firmware it's ok, with a commit new file becomes tested.
Wen rollback occour diagnostic log aid to investigate the problem if possible memorize it to sdcard file.

pat1, 2016-04-29:
I firmware modificati dovrebbero andare qui:
https://github.com/r-map/rmap/tree/master/arduino/sketchbook/hardware/Microduino/avr/bootloaders

2boots pare un bootloader un po' vecchio.
Lo stato dell'arte parrebbe essere optiboot:
https://github.com/arduino/Arduino/tree/master/hardware/arduino/avr/bootloaders

Una versione modificata di optiboot è:
https://github.com/Rodot/Gamebuino/tree/master/hardware/gamebuino/bootloaders/gamebuino_boot
http://gamebuino.com/wiki/index.php?title=Bootloader
probabilmente è da qui che bisogna partire ...

pat1, 2016-04-29:
link utili per i pn da utilizzare su microduino per ISP:
https://www.arduino.cc/en/Tutorial/ArduinoISP
http://www.instructables.com/id/Adding-ICSP-header-to-your-ArduinoAVR-board/step2/Finding-the-pins-to-connect-to/
https://wiki.microduino.cc/index.php/Microduino_Upin27_Pinout_Summary

quindi su microduino i pin sarebbero:
miso  -> D11
mosi -> D12
sck -> D13
rst -> RST