r-map/rmap #40: how we ca power supply miroduino PoE ?

pat1, 2015-10-02:
[Microduino-RJ45.pdf](https://github.com/r-map/rmap/files/6915/Microduino-RJ45.pdf)
Ho posto questa questione sul forum di microduino:
https://www.microduino.cc/forum/topic/72/power-over-ethernet-with-microduino-enc28j60

considerando questo schema elettrico:
https://www.microduino.cc/wiki/images/f/fc/Microduino-RJ45.zip
avete suggerimenti ?

Viste le caratteristiche del LM2674
http://www.farnell.com/datasheets/10234.pdf
parrebbe che 24 V siano la tensione giusta, ma da appllicare dove e
come ?

Allego lo schema in PDF

pat1, 2015-10-12:
I suppose that the module require
10/100 mode B,DC on spares with 24V
so the http://dlnmh9ip6v2uc.cloudfront.net/datasheets/Cables/Ethernet/poedatadsheet.pdf cable have to work without problems.

closed 2015-10-12 by pat1

pat1, 2015-10-12:
e funziona.

dcast78, 2015-10-12:
ottimo!