r-map/rmap #49: server: verificare l'esportazione geojson in arkiweb

pat1, 2015-10-15:


pat1, 2015-11-13:
Bisogna cambiare il plugin per il nuovo formato json prodotto da dballe

edigiacomo, 2015-11-13:
Per retrocompatibilità, userei un nome diverso di plugin. Essendo stata una scelta infelice quella di chiamarlo `json` (ero troppo distratto dai miei capelli giovanili, al tempo), non starei a perseverare.

Propongo quindi di fare un plugin `dbajson`. Il vecchio plugin `json` secondo me bisogna decidere se dismetterlo oppure se rinominarlo in `geojson`.

Comunque, direi di githubbare anche i plugin di arkimet, in cui sposterò questa issue (e a quel punto chiuderò questa).

edigiacomo, 2015-11-16:
I postprocessatori `json` e `singlepoint` in [arpa-simc/arkimet-postprocessor-suite](https://github.com/ARPA-SIMC/arkimet-postprocessor-suite/) usano, tra gli altri, anche il formato `dbajson`.

closed 2015-11-16 by edigiacomo