r-map/rmap #74: no matching function for call to 'PubSubClient::PubSubClient(char [8], int, void (&)(char*, byte*, unsigned int), sim800Client&)'

dmsherazi, 2016-01-13:
I am trying to compile sim800/mqtt_basic.ino and am getting this error

no matching function for call to 'PubSubClient::PubSubClient(char [8], int, void (&)(char*, byte*, unsigned int), sim800Client&)'

pat1, 2016-01-13:
In:
 libraries/PubSubClient/PubSubClient.h
you have to use those definitions:

 // if use sim800 client
 #include "sim800Client.h"
 #define TCPCLIENT sim800Client

 // if use arduino_uip or etherclient
 //#include "Client.h"
 //#include "Stream.h"
 //#define TCPCLIENT Client

dmsherazi, 2016-01-13:
I did change the define and include requirements. I am still getting this error.

`````

mqtt_basic:38: error: no matching function for call to 'PubSubClient::PubSubClient(char [8], int, void (&)(char*, byte*, unsigned int), sim800Client&)'

 PubSubClient client(server, 1883, callback, s800);```

pat1, 2016-01-13:
Are you sure that you are using rmap specific PubSubClient and not the main stream standard library? The main stream standard library cannot work with sim800 library.

dmsherazi, 2016-01-14:
I guess there was an issue with arduino IDE not rebuilding the library files and using old .o files. Problem solved after forcing the IDE to rebuild files by changing the board

closed 2016-01-14 by pat1