r-map/rmap #77: mqtt connection messages go in dballe DB

pat1, 2016-01-17:
001213 AIRBASE AIR QUALITY OBSERVING STATION CODE(CCITTIA5): conn

edigiacomo, 2016-01-17:
@pat1 `mqtt2bufr` considers only the topics with a pattern `rmap/IDENT/COORDS/NETWORK/-,-,-/-,-,-,-/VAR`.  So, `rmap/IDENT/COORDS/NETWORK/-,-,-/-,-,-,-/B01213` is considered as a constant data.

Reading the [rmap specs](http://www.raspibo.org/wiki/index.php/Gruppo_Meteo/RFC-rmap#Constant_Data), it seems that constant data are sent all together with the topic `rmap/IDENT/COORDS/NETWORK/-,-,-/-,-,-,-` (I don't recall having read this part of the specs when I implemented `mqtt2bufr`). **Note**: the specs don't specify where the `var` values are set (maybe in the payload?).

Are you sure that your clients are sending the constant data with the `rmap/IDENT/COORDS/NETWORK/-,-,-/-,-,-,-/` pattern (without `VAR`)? In this case, `mqtt2bufr` would ignore (throwing an exception) all the constant data. If you find some constant data in your Db-All.e database then your client is sending the constant data specifying the `VAR` in the topic and this means that there's an ambiguity in your client (`rmap/IDENT/COORDS/NETWORK/-,-,-/-,-,-,-/B01213` is a constant data or a connection state?).

If I understand correctly, `mqtt2bufr` should parse 3 kind of topics:

* `rmap/IDENT/COORDS/NETWORK/-,-,-/-,-,-,-/B01213`: special case for the state of the connection.
* `rmap/IDENT/COORDS/NETWORK/-,-,-/-,-,-,-`: all the constant data (**note**: the specs don't say anything about where to find the values of the vars).
* Any other pattern `rmap/IDENT/COORDS/NETWORK/TRANGE/LEVEL/VAR` (but with `TRANGE != -,-,-` and `LEVEL != -,-,-,-`) is a time-dependent value.

pat1, 2016-01-18:
you are right ;)
is possible this is a feature and not a bug