r-map/rmap #79: v7d_transform do not create report any more

pat1, 2016-01-17:
the command:
v7d_transform --input-format=dba --output-format=BUFR --network-list=rmap --start-date "2016-01-17 10:13:40" --end-date "2016-01-17 23:15:00" --comp-start "2016-01-17 10:13:40" --comp-step "0000000000 00:01:00.000" --comp-frac-valid .002 --comp-stat-proc=254:0 --set-network generic  rmap/rmap@rmap -

have no output

dcesari, 2016-01-18:
Can you attach an input example?
Which is the frequency of input data?
longest=comp-step/(comp-frac-valid*999  +1)=~0.33*comp-step=~20''

pat1, 2016-01-18:
frequency of input data should be 5 sec.
[rmap.zip](https://github.com/r-map/rmap/files/94038/rmap.zip)

pat1, 2016-01-18:
il problema è di dballe che nelle api fortran non scrive più in stdout quando il nomefile è "".

pat1, 2016-01-28:
il problema era in libsim; risolto

closed 2016-01-28 by pat1