r-map/rmap #81: mqtt2bufr: input missing level, output wrong level

pat1, 2016-01-18:
input level:
0,7,-,-
diventa in output level:
0,7,0,0
https://github.com/ARPA-SIMC/libsim/issues/11

mosquitto_sub -h rmap.cc -v -t rmap/dancast78/\#
rmap/dancast78/1064555,4469950/rmap/-,-,-/-,-,-,-/B01213 {"v":"conn"}
rmap/dancast78/1064555,4469950/rmap/254,0,0/0,7,-,-/B12101 {"v":31821}
rmap/dancast78/1064555,4469950/rmap/254,0,0/0,7,-,-/B12101 {"v":31821}
rmap/dancast78/1064555,4469950/rmap/254,0,0/0,7,-,-/B12101 {"v":31821}

mqtt2bufr -t rmap/dancast78/\# |dbamsg dump --interpreted
#9[0] generic message with 2 contexts:
Level 0,7,0,0  tr 254,0,0  2 vars:
001213 AIRBASE AIR QUALITY OBSERVING STATION CODE(CCITTIA5): conn
012101 TEMPERATURE/DRY-BULB TEMPERATURE(K): 318.21
Level -,-,-,-  tr -,-,-  4 vars:
001011 SHIP OR MOBILE LAND STATION IDENTIFIER(CCITTIA5): dancast78
001194 [SIM] Report mnemonic(CCITTIA5): rmap
005001 LATITUDE (HIGH ACCURACY)(DEGREE): 44.69950
006001 LONGITUDE (HIGH ACCURACY)(DEGREE): 10.64555

edigiacomo, 2016-01-18:
Credo sia un bug in dballe, che intepreta la stringa `-` su livello e timerange come 0 (zero) invece che come missing: https://github.com/ARPA-SIMC/dballe/issues/29.

edigiacomo, 2016-01-19:
Reopening the bug, I will close it when the ARPA-SIMC/dballe#29 will be fixed (or rejected).

pat1, 2016-01-28:
```
mosquitto_sub -v -t rmap/pat1/\#
rmap/pat1/1131844,4449446/rmap/-,-,-/-,-,-,-/B01213 {"v":"conn"}
rmap/pat1/1131844,4449446/rmap/254,0,0/0,1,-,-/B12101 {"v":30690,"t":"2016-01-28T11:22:53"}
rmap/pat1/1131844,4449446/rmap/254,0,0/0,1,-,-/B12101 {"v":30690,"t":"2016-01-28T11:22:58"}

 mqtt2bufr -t rmap/pat1/\# | dbamsg dump 
#0 BUFR message: 148 bytes, origin 200:0, category 0 255:255:0, bufr edition 4, tables 14:1, subsets 1, values: 18/18:
Subset 0:
001194 [SIM] Report mnemonic(CCITTIA5): rmap
004001 YEAR(YEAR): 2016
004002 MONTH(MONTH): 1
004003 DAY(DAY): 28
004004 HOUR(HOUR): 11
004005 MINUTE(MINUTE): 24
004006 SECOND(SECOND): 55
001011 SHIP OR MOBILE LAND STATION IDENTIFIER(CCITTIA5): pat1
005001 LATITUDE (HIGH ACCURACY)(DEGREE): 44.49446
006001 LONGITUDE (HIGH ACCURACY)(DEGREE): 11.31844
007192 [SIM] First level type(NUMERIC): 0
007193 [SIM] Level L1(NUMERIC): 0
007195 [SIM] Second level type(NUMERIC): 0
007194 [SIM] Level L2(NUMERIC): 0
004192 [SIM] Time range type(NUMERIC): 0
004193 [SIM] Time range P1(NUMERIC): 0
004194 [SIM] Time range P2(NUMERIC): 0
001213 AIRBASE AIR QUALITY OBSERVING STATION CODE(CCITTIA5): conn
#1 BUFR message: 144 bytes, origin 200:0, category 0 255:255:0, bufr edition 4, tables 14:1, subsets 1, values: 18/18:
Subset 0:
001194 [SIM] Report mnemonic(CCITTIA5): rmap
004001 YEAR(YEAR): 2016
004002 MONTH(MONTH): 1
004003 DAY(DAY): 28
004004 HOUR(HOUR): 11
004005 MINUTE(MINUTE): 24
004006 SECOND(SECOND): 53
001011 SHIP OR MOBILE LAND STATION IDENTIFIER(CCITTIA5): pat1
005001 LATITUDE (HIGH ACCURACY)(DEGREE): 44.49446
006001 LONGITUDE (HIGH ACCURACY)(DEGREE): 11.31844
007192 [SIM] First level type(NUMERIC): 0
007193 [SIM] Level L1(NUMERIC): 1
007195 [SIM] Second level type(NUMERIC): 0
007194 [SIM] Level L2(NUMERIC): 0
004192 [SIM] Time range type(NUMERIC): 254
004193 [SIM] Time range P1(NUMERIC): 0
004194 [SIM] Time range P2(NUMERIC): 0
012101 TEMPERATURE/DRY-BULB TEMPERATURE(K): 306.90
```

commit 0958413e (2016-01-28): Fix #81
commit c0b5aefe (2016-01-29): Revert "Fix #81"

This reverts commit 0958413ec483abe683929dfd7b8cfabb413f6cc1.
closed 2016-01-28 by edigiacomo