r-map/rmap #89: server: main menu (use js) do not work outside borinud

pat1, 2016-01-27:
js error due to jquery reload inside borinud.

pat1, 2016-01-27:
eliminato jquery da borinud ma forse c'è un metodo migliore?

edigiacomo, 2016-01-28:
No, in effetti eliminare jQuery da Borinud ha senso, visto che per ora non è una app portabile ma i templates (e quindi anche l'importazione di librerie JavaScript) sono del project.

Direi di chiudere questa issue e di aprire una richiesta di enhancement per far diventare Borinud una app riutilizzabile (e quindi gestire per bene l'inclusione di librerie JavaScript).

Non so quali siano le best practices nel caso di app riutilizzabili che fanno uso di librerie JavaScript che potrebbero essere già inclusi nel project (come nel caso di jQuery). La soluzione più semplice che mi viene in mente è quella di specificare, nella doc della app, che si devono includere certe librerie JavaScript (e.g. jQuery > vX.Y).

closed 2016-01-28 by pat1