r-map/rmap #93: mqtt2bufr: do not write a right bufr on constant station data

pat1, 2016-01-29:
./mqtt2bufr -t rmap/dancast78/\# |dbamsg dump 
#0 BUFR message: 112 bytes, origin 200:0, category 0 255:255:0, bufr edition 4, tables 14:1, subsets 1, values: 5/11:
Subset 0:
001194 [SIM] Report mnemonic(CCITTIA5): rmap
004001 YEAR(YEAR): (undef)
004002 MONTH(MONTH): (undef)
004003 DAY(DAY): (undef)
004004 HOUR(HOUR): (undef)
004005 MINUTE(MINUTE): (undef)
004006 SECOND(SECOND): (undef)
001011 SHIP OR MOBILE LAND STATION IDENTIFIER(CCITTIA5): dancast78
001213 AIRBASE AIR QUALITY OBSERVING STATION CODE(CCITTIA5): conn
005001 LATITUDE (HIGH ACCURACY)(DEGREE): 44.69950
006001 LONGITUDE (HIGH ACCURACY)(DEGREE): 10.64555

messaggio che non può essere importato con dbadb import

pat1, 2016-01-29:
Da fortran io utilizzo indirettamente
common api:
virtual void setcontextana();

che sarebbe:
void CommonAPIImplementation::setcontextana()
{
    input.set_datetime(Datetime());
    input.set_level(Level());
    input.set_trange(Trange());
    station_context = true;
}

ne segue una domanda:
perchè non usare le commonapi?
forse perchè sono in stile vecchio fortran ?

pat1, 2016-01-29:
parrebbe un bug di dballe in quanto anche con le api fortran produco un file con solo dati di anagrafica che produce lo stesso errore.
dbadb import --wipe-first --dsn=sqlite:/tmp/test.sqlite  example_dballe_ana.bufr
date/time informations not found (or incomplete) in message to insert

dbamsg dump example_dballe_ana.bufr
#0 BUFR message: 92 bytes, origin 200:0, category 0 255:255:0, bufr edition 4, tables 14:1, subsets 1, values: 2/9:
Subset 0:
001194 [SIM] Report mnemonic(CCITTIA5): (undef)
004001 YEAR(YEAR): (undef)
004002 MONTH(MONTH): (undef)
004003 DAY(DAY): (undef)
004004 HOUR(HOUR): (undef)
004005 MINUTE(MINUTE): (undef)
004006 SECOND(SECOND): (undef)
005001 LATITUDE (HIGH ACCURACY)(DEGREE): 45.00000
006001 LONGITUDE (HIGH ACCURACY)(DEGREE): 13.00000

edigiacomo, 2016-01-29:
Se in dballe non è possibile importare messaggi del genere, allora si deve capire se è un bug oppure un limite (e quindi aprire una issue su dballe come bug o enhancement).

Dopo che l'hai aperta, lascia riferimento qui, così non appena su dballe è chiusa la possiamo chiudere anche qui.

pat1, 2016-03-18:
per ora risolto con c65cf49