r-map/rmap #96: app: picture - using front camera picture results in a black photo

brancomat, 2016-02-03:
(maybe resolution related?)
with rear camera (13MP) shows a black block instead of picture taken
with front camera (8MP) the picture is fine

pat1, 2016-02-11:
puoi allegare qui una immagine with rear camera (13MP)

brancomat, 2016-02-11:
eccallà

[image]

pat1, 2016-02-11:
'''
In [4]: jpeg = pexif.JpegFile.fromFile("test.jpg")
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-4-5af936479456> in <module>()
----> 1 jpeg = pexif.JpegFile.fromFile("test.jpg")

/home/pat1/git/rmap/python/rmap/pexif.pyc in fromFile(filename, mode)
   1001         """Return a new JpegFile object from a given filename."""
   1002         with open(filename, "rb") as f:
-> 1003             return JpegFile(f, filename=filename, mode=mode)
   1004     fromFile = staticmethod(fromFile)
   1005 

/home/pat1/git/rmap/python/rmap/pexif.pyc in __init__(self, input, filename, mode)
   1064                     # Note: Segment class may modify the input file
   1065                     # descriptor. This is expected.
-> 1066                     attempt = segment_class(mark, input, data, self.mode)
   1067                     segments.append(attempt)
   1068                     break

/home/pat1/git/rmap/python/rmap/pexif.pyc in __init__(self, marker, fd, data, mode)
    857         self.e = '<'
    858         self.tiff_endian = 'II'
--> 859         DefaultSegment.__init__(self, marker, fd, data, mode)
    860 
    861     def parse_data(self, data):

/home/pat1/git/rmap/python/rmap/pexif.pyc in __init__(self, marker, fd, data, mode)
    149         assert mode in ["rw", "ro"]
    150         if self.data is not None:
--> 151             self.parse_data(data)
    152 
    153     class InvalidSegment(Exception):

/home/pat1/git/rmap/python/rmap/pexif.pyc in parse_data(self, data)
    901             start = 2 + offset + (num_entries*12)
    902             if (count == 1):
--> 903                 ifd = IfdTIFF(self.e, offset, self, self.mode, tiff_data)
    904             elif (count == 2):
    905                 ifd = IfdThumbnail(self.e, offset, self, self.mode, tiff_data)

/home/pat1/git/rmap/python/rmap/pexif.pyc in __init__(self, e, offset, exif_file, mode, data)
    417             if tag in self.embedded_tags:
    418                 try:
--> 419                     actual_data = self.embedded_tags[tag][1](e, the_data, exif_file, self.mode, data)
    420                 except JpegFile.SkipTag as exc:
    421                     # If the tag couldn't be parsed, and raised 'SkipTag'

/home/pat1/git/rmap/python/rmap/pexif.pyc in __init__(self, e, offset, exif_file, mode, data)
    417             if tag in self.embedded_tags:
    418                 try:
--> 419                     actual_data = self.embedded_tags[tag][1](e, the_data, exif_file, self.mode, data)
    420                 except JpegFile.SkipTag as exc:
    421                     # If the tag couldn't be parsed, and raised 'SkipTag'

/home/pat1/git/rmap/python/rmap/pexif.pyc in ifd_maker_note(e, offset, exif_file, mode, data)
    622 def ifd_maker_note(e, offset, exif_file, mode, data):
    623     """Factory function for creating MakeNote entries"""
--> 624     if exif_file.make == "Canon":
    625         # Canon maker note appears to always be in Little-Endian
    626         return CanonIFD('<', offset, exif_file, mode, data)

AttributeError: ExifSegment instance has no attribute 'make'
'''

pat1, 2016-02-11:
App da provare:
Link (Will expire in 30 days): http://www.comodino.org/bytehoard/filelink.php?filecode=e1d9191c196280451cb0ec36ca8b6055457c03afdbe3e0886d948a7967883f74

brancomat, 2016-03-31:
it seems to be fixed in 6.10

commit bde397cf (2016-02-11): do not fault on missed maker; close #96
closed 2016-03-31 by brancomat