r-map/rmap #106: server: support Arkimet BUFR in Borinud

edigiacomo, 2016-02-25:
Support Arkimet BUFR in Borinud.

edigiacomo, 2016-02-25:
@pat1 is this issue part of the Milestone *server rmap release 2*?

pat1, 2016-02-25:
si mettiamola !

edigiacomo, 2016-02-25:
Ho iniziato il lavoro in f7d7d76 ma c'è ancora un po' da fare:
- L'elenco delle terne `var`, `level`, `trange` è predefinito per tutto il dataset: è come se tutte le stazioni misurassero sempre le stesse variabili (una sorta di template). Mi rendo conto che è una grossa approssimazione, ma sarebbe già qualcosa.
- La query delle stazioni non restituisce nessun dato statico: è come se le stazioni non ne avessero, si deve capire come tenere traccia di dati statici di stazione.
- Da integrare nella configurazione `BORINUD["SOURCES"]`: una volta che è possibile specificare questo tipo di database da `settings.py`, direi di testarne il funzionamento, chiudere questa issue e crearne due (una per ognuno dei due punti precedenti), così possiamo discutere come muoverci e fin dove arrivare per completare la milestone.

commit 9a4f8f0c (2016-02-26): ArkimetBufrDB.query_data yields records

Fix #106
PR #107 (2016-02-26): Fix #106
Questa PR permette una fix #106 parziale. [Ancora un po' di cose rimangono da fare](https://github.com/r-map/rmap/issues/106#issuecomment-188924983) e per quelle aprirò delle issue nuove.

Attenzione @pat1: `BORINUD["SOURCES"]` ha cambiato sintassi! Ti lascio l'onore di fare il merge di questa PR :)
commit 7064565f (2016-02-26): Merge pull request #107 from r-map/issue/106

Fix #106
closed 2016-02-26 by pat1