r-map/rmap #119: server: mqtt2bufr freeze

pat1, 2016-03-14:
mqtt2bufr freeze 
try with apache restart

edigiacomo, 2016-03-14:
Scusa, ma non capisco la issue. Cosa c'entra mqtt2bufr con Apache? E in che modo si "congela"?

pat1, 2016-03-14:
scusa, ho sbagliato tutto, era più che altro una nota per non perdermelo.
bufr2mqtt
il freeze è possibile che avvenga quando si ferma apache.
Apache fornisce il servizio http per l'autenticazione di mosquitto.
E' possibile che questo generi un errore e faccia andare in loop infinito bufr2mqtt.

edigiacomo, 2016-03-14:
Nessun problema, mi mancava il pezzo dell'autenticazione via Apache per Mosquitto.

Ci guardiamo a quattro occhi, allora, non appena ci incrociamo al lavoro.

pat1, 2016-03-18:
closed by d8ccb20

closed 2016-03-18 by pat1