r-map/rmap #125: help- how to issue other commands  to sim800 while still being connected to mqtt server

dmsherazi, 2016-03-21:
If i need to check sms or signal strength while connected to mqtt serevr on sim800 I have to issue escape sequence and this will disconnect me from serevr! 

I need to check for incoming calls and sms and at times for signal strength or CREG . I haven't used sim800 a lot for data before so my pardon me if its a silly question!

pat1, 2016-03-21:
This library implement sim800 for transparent tcp connection and do other things when transparent is active is not possible.

But after a switch Between Data Mode to Command Mode The manual report:
```
ATO command can be used to switch the module from command mode to data mode if the
connection is active, and CONNECT will be returned again. 
```
so you can try :
```
sim800Client::transparentescape()
.... do somethings
....
sim800Client::transparent()
```
but the time from transparentescape and transparent cannot exceed :
```
// MQTT_KEEPALIVE : keepAlive interval in Seconds
#define MQTT_KEEPALIVE 30
```
in PubSubClient.h
Please report here if this works.

There are other modes to do tcp connection with sim800; multi connection mode
```
Multi Connection
To enable multi connection mode, the command AT+CIPMUX must be set to 1. Be aware
that in the multi connection mode, SIM800 series only supports non-transparent mode. Due to this
matter, the command AT+CIPMODE is invalid under multi connection mode.
In multi connection mode, SIM800 series can work as a TCP/UDP client or a TCP server.
As a client, it can establish 6 connections to remote server in total, both TCP and UDP. 
```
Multi connection is not implemented in this library.

For other functionality take a look at SIM800 Series_TCPIP_Application 
```
2.2.5
How to handle incoming call and short message in data mode 
```

dmsherazi, 2016-03-22:
Thanks a lot @pat1 for the detailed answer.

closed 2016-03-22 by dmsherazi