r-map/rmap #126: Can we use ssl  with PubSubClient for arduino?

dmsherazi, 2016-03-21:


pat1, 2016-03-21:
For what I know there are not ssh implementation over small microcontroller.

closed 2016-03-21 by pat1

ppYodiwo, 2016-11-01:
SIM800 module does have a basic http ssl implementation in order to connect to a ssl secured mqtt server.

dmsherazi, 2016-11-02:
@ppYodiwo can you share the resource or any documentation for ssl in sim800/sim900

ppYodiwo, 2016-11-02:
Apologies for my short comment. 
I did not elaborate since the issue was already a few months old and closed.
Some more details:
Theory:
Both SIM900 and SIM800 have SSL support for https and ftp depending on the firmware of the module.
This allows the modules to perform htpps get and post  for example.
Documentation can be found in: http://www.rhydolabz.com/documents/sim900_https_at_command_set_v1_00.pdf

MQTT is on TCP layer therefor the https is not useful.
SIM800 though also supports SSL on TCP level.
Documentation can be found in:
http://mt-system.ru/sites/default/files/documents/sim800_series_ssl_application_note_v1.01.pdf

Practice:
I have used your library (great job by the way) in one of my projects with slight modifications in order to enable the SSL functionality.
I can confirm that I am able to subscribe to a SSL mqtt server on port 8883 receive messages and publish. (Some issues I have are out of scope of this message)

If you want I can send you my changes if you think this could be a usefull addition to your library.

If you need any more specification please ask away.

dmsherazi, 2016-11-03:
@ppYodiwo Thanks for the details. It would be very nice if you can make PR with your changes. Additionally if you guide which firmware supports SSL and share the firmware file if possible as well.
Regards

ppYodiwo, 2016-11-03:
Regarding the firmware version:
The card I have has the following revision number:
Revision: 1308B02SIM800M64_BT
You can find the revision number by issung the command AT+CMGR.
My card had this firmware from scratch so I do not know about the firmware upgrade procedure. As for firmware images a number of them can be found in: 
 http://dostmuhammad.com/blog/sim900-firmware-update-tutorials-appnotes/  
8-)
As for the PR I will try to do it over the weekend. (I need to find out how to do the PR since I am not very familiar with github procedures)