r-map/rmap #161: Adattamento Amatyr ai json rmap

EnricoScariot, 2016-06-29:
Buongiorno, sono un tirocinante che era stato incaricato di sviluppare un pannello per la visualizzazione in tempo reale dei dati di una stazione di monitoraggio meteo (attraverso l'utilizzo dei json provenienti da rmap) facendo uso del software Amatyr.
Ho lavorato alle sezioni dei grafici relative a 3day, today, yesterday, month e year inserendo come parametri: temperatura, umidità, pressione, pioggia. Nel caso di month ho aggiunto i parametri temperatura massima e temperatura minima poiché le rilevazioni sono giornaliere. 
Il codice presenta dei bug, infatti ogni tanto bisogna ricaricare la pagina affinché il grafico venga visualizzato nel modo corretto. Questo, secondo il mio punto di vista, è dovuto all'utilizzo della funzione d3.json che prende come input un url e ne restituisce il json in esso contenuto. Questo problema nel software originale non è presente dal momento che per la realizzazione del grafico di ogni sezione (mensile o giornaliera) viene fatta una sola chiamata a quella funzione mentre nel nostro caso per recuperare tutte le info necessarie devo farne una per parametro (temperatura, pressione, umidità e pioggia). Questo crea problemi perché (suppongo io) se per caso c'è un ritardo nel captare il json dall'url il programma esegue la prossima chiamata alla d3.json che prevederebbe che il json sia già stato inizializzato.
Un'altra cosa a me incomprensibile è che la funzione che genera i grafici chiamata draw e dichiarata all'interno di amatyrlib.js funziona in modo corretto solo se viene chiamata all'interno di una delle funzioni d3.json altrimenti il grafico viene generato solo con lo strumento per sviluppatori di Google Chrome aperto. 
Potete visualizzare le modifiche nel branch da me creato "AmatyrEnrico2"
Distinti Saluti
Enrico Scariot

pat1, 2016-06-30:
Io ottengo questi errori con firefox e forse la cosa non aiuta:
```
TypeError: imgElem is null
newImg.src = imgElem.src.split("?")[0] + "?" + new Date().getTime();
amatyr.js (linea 744, col 6)

ReferenceError: current_weather is not defined
for(key in current_weather.current) {
amatyr.js (riga 493)

```

pat1, 2016-06-30:
Ecco le istruzioni per installare il sofware server di rmap ed effuare i test:

```
 git clone https://github.com/r-map/rmap.git rmap
 cd rmap
 git checkout AmatyrEnrico2
 cd python
 virtualenv --no-site-packages myvenv
 .  myvenv/bin/activate
 mkdir  -p myvenv/share/rmap
 
 pip install django
 pip install ConfigObj
 pip install django-leaflet
 pip install django-geojson
 pip install django-imagekit
 pip install pillow
 pip install jsonfield
 pip install pika
 pip install simplejson
 
 ./rmapctrl --syncdb
 python manage.py migrate
 ./rmapweb
```
Aggiungo che per fare delle prove bisogna installare CORS everywhere in
firefox:

 https://addons.mozilla.org/en-US/firefox/addon/cors-everywhere/
This is a firefox addon that allows the user to enable CORS everywhere
by altering http responses.

attivarlo con :
The addon's functionality can be toggled with the included button and is
disabled by default.
The button can be found by right-clicking a toolbar and choosing
`customize`.
It is labelled `CorsE` with a red background that turns green when
enabled.

Poi puntare il browser a http://127.0.0.1:8888/amatyr
e selezionare il bottone Today

NonnEmilia, 2016-06-30:
Gli errori segnalati da @pat1 sono legati ai task che si ripetono ogni 30/60 secondi.

#### Auto update webcam

    TypeError: imgElem is null

esce perché manca nell'HTML un tag `<img/>` con `id="webcam"` in cui andare a inserire l'immagine della webcam.

#### Fetch current weather with 1 minute interval

    ReferenceError: current_weather is not defined

esce perché la variabile `current_weather` non è ancora stata inizializzata nel punto in cui viene chiamata.

pat1, 2016-07-06:
Riporto dalla pull request:
Il problema delle più chiamate da aspettare è più comodo da risolvere con jQuery piuttosto che con d3.
Come hai giustamente notato tu bisogna attendere che tutte quante le chiamate abbiano restituito il JSON prima di poter graficare.

La funzione $.when() si aspetta come argomenti una serie di chiamate $.ajax() e chiama la callback passata in .done() solo quando tutte le chiamate sono terminate con successo.

In questo modo si rendono nuovamente sincrone una serie di chiamate asincrone e riesci a chiamare draw() quando sei sicuro di avere tutti i dati.

Partendo da quell'esempio si possono correggere correggere tutte le chiamate di questa issues