r-map/rmap #181: Specs for Borinud V2 API format

edigiacomo, 2016-12-14:
We should write the specs for v2 APIs and decide which format will be supported:

- JSON
- JSON-stream
- JSON *and* JSON-stream

pat1, 2016-12-14:
ho fatto un po' di casino ...
ci siamo sovrapposti con i commit lavorando sullo stesso file contemporaneamente ossia 
python/borinud/v2/views.py 

ho provato a fare un merge, ma bisogna riguardarci accuratamente ....

https://github.com/r-map/rmap/commit/40b53570fc693a80f6d0f7cf38394f4d7470a328#diff-c2cc94971386397e93df732221a4df70

edigiacomo, 2016-12-14:
Niente di grave (anche perché attualmente le API v2 non sono usate): ti propongo di scrivere le specifiche delle API v2 e poi ci organizziamo sullo sviluppo, che ne dici?

edigiacomo, 2017-01-18:
@pat1 chiudo questa issue perché hai tolto la versione 2 delle API e ampliato la versione 1 con nuovi formati. Apro però la #193, perché il problema delle specifiche rimane.

closed 2017-01-18 by edigiacomo