r-map/rmap #206: mqtt2bufr: strange value when invalida data in payload

pat1, 2017-03-24:
to reproduce:

mosquitto_pub -h rmap.cc -t test/dancast78/1137651,4449231/rmap/254,0,0/103,2000,-,-/B15198 -m '{"t": "2017-03-24T06:18:10", "v": "None"}'

"None" here is invalid as numeric value for "v" (should be null)

mqtt2bufr -h rmap.cc -t test/dancast78/\#|dbamsg dump --interpreted 
#0[0] generic message, rep_memo: , 44.49231,11.37651, ident: dancast78, dt: 2017-03-24T06:18:10, 2 contexts:
Level 103,2000,-,-, tr 254,0,0 1 vars:
015198 [SIM] PM2.5 Concentration(KG/M**3): 0.0000036973
Level -,-,-,-, tr -,-,-, 4 vars:
001011 SHIP OR MOBILE LAND STATION IDENTIFIER(CCITTIA5): dancast78
001194 [SIM] Report mnemonic(CCITTIA5): rmap
005001 LATITUDE (HIGH ACCURACY)(DEGREE): 44.49231
006001 LONGITUDE (HIGH ACCURACY)(DEGREE): 11.37651

36973 have nothing to say here ...

edigiacomo, 2017-03-28:
`mqtt2bufr` delegates the conversion to the overloaded function `wreport::Var::set` ([code](https://github.com/r-map/rmap/blob/master/mqtt2bufr/parser.cc#L148-L156)). When the var is numeric,  `wreport` converts the string to integer ([code](https://github.com/ARPA-SIMC/wreport/blob/master/wreport/var.cc#L496-L510)).

Maybe this is a `wreport` issue?

pat1, 2021-04-30:
solved in wreport

closed 2021-04-30 by pat1