r-map/rmap #218: Arduino interface with Sim800 using MQTT protocol

priya2212, 2017-06-19:
Hello everyone,
I am using this library to run MQTT over Sim800 module. I am using Arduino UNO to interface with SIm800 module. So, to disable the serial1 port, I comment out this line on sim800.h:

> //#define DEBUGONSERIAL
> #define ENABLEWDT 1
> // if define HARDWARESERIAL is set then it will be static (required by Time library)
> **//#define HARDWARESERIAL Serial1**

But it is giving me error in my main file:
error: 'Serial1' was not declared in this scope
     if (s800.init(&Serial1 , 7, 6)) break;
I want to use software serial for my UNO, It would be great for me if anyone can guide me for how to use software serial with this library because whenever I am disabling the hardware serial, it is throwing a lot of errors in the library. 
Thanks & Regards,
Priya

pat1, 2017-06-19:
Take a look here:
https://github.com/r-map/rmap/issues/202

also you have to change in 
https://github.com/r-map/rmap/blob/master/arduino/sketchbook/libraries/PubSubClient/PubSubClient.h
uncomment:
#include "sim800Client.h"
#define TCPCLIENT sim800Client

and commet:
//#include "Client.h"
//#include "Stream.h"
//#define TCPCLIENT Client

priya2212, 2017-06-20:
Hi @pat1, Thank you so much for your response. I have uncommented and commented the above required stuff. It is still throwing me the error. & And I have gone through the issue #202 for using serial of Arduino UNO. but I'm not getting any response at the end of other client when I'm subscribing to the topic. Even the module is not initiating connection with TCP/IP as well and I am not able to debug it as I have disabled the serial.

pat1, 2017-06-21:
> Hi @pat1, Thank you so much for your response. I have uncommented and commented the above required stuff. It is still throwing me the error. 

which error?

> And I have gone through the issue #202 for using serial of Arduino UNO. but I'm not getting any response at the end of other client when I'm subscribing to the topic. Even the module is not initiating connection with TCP/IP as well 

how do you know this? what are you monitoring?

> and I am not able to debug it as I have disabled the serial.

This is why I use 644p or 1284P (arduino mega) to devel ; try to use software serial for debug connected to something you can read from .  Or you can try to use software serial for sim800 but for me this is not a good solution.

closed 2020-07-10 by pat1