r-map/rmap #220: substitute aJson with ArduinoJson

pat1, 2017-07-25:
Questione che blocca la migrazione a ArduinoJson:
ArduinoJson è in grado di sostituire un uso un po' articolato
di aJSON come si fa in questo firmware ad esempio:
https://github.com/r-map/rmap/blob/master/arduino/sketchbook/libraries/RadioHead/examples/cc11/jsrpc/jsrpc.ino
dove si fa uso di api un po' particolari tipo:
aJson.deleteItemFromObject
aJson.detachItemFromObject

mbaldinetti, 2017-07-27:
le due funzioni elencate non sono disponibili nella libreria indicata per la scelta dell'allocazione statica della stessa libreria.

guardando la libreria aJSON ho notato che usa della malloc: https://github.com/r-map/rmap/blob/master/arduino/sketchbook/libraries/aJson/aJSON.cpp#L190 

se non ho capito male, esiste la possibilità di usare un buffer statico?

pat1, 2017-07-27:
https://github.com/r-map/rmap/blob/master/arduino/sketchbook/libraries/aJson/aJSON.cpp#L617

pat1, 2017-07-27:
molto interessante,
quella malloc era assolutamente sfuggita all'analisi.

L'eliminazione della malloc a cui mi riferivo è:
https://github.com/r-map/rmap/blob/master/arduino/sketchbook/libraries/aJson/aJSON.cpp#L617
che sotituisce volendo:
https://github.com/r-map/rmap/blob/master/arduino/sketchbook/libraries/aJson/aJSON.cpp#L602

Quindi le mie affermazioni si sono rivelate false e aJSON fa un uso intesivo di malloc e free
le possibilità rimangono tre:
* accertarsi che ogni malloc sia seguita dalla sua corrispondente free (cosa effettivamente difficile da certificare)
* implementare in aJSON l'allocazione da buffer statico come in https://github.com/bblanchon/ArduinoJson/blob/master/src/ArduinoJson/StaticJsonBuffer.hpp
* migrare a ArduinoJson (che pare a questo punto la soluzione più semplice)

pat1, 2017-07-27:
segnalo solo in appendice l'uso di https://github.com/bblanchon/ArduinoJson/blob/master/src/ArduinoJson/StaticJsonBuffer.hpp richiede un po' di analisi;  il suo funzionamento pare molto simile alla gestione della malloc solo dedicata alla singola libreria; non so se possa creare gli stessi problemi di frammentazione...

mbaldinetti, 2017-07-28:
concordo con l'analisi.

da quello che leggo in https://bblanchon.github.io/ArduinoJson/doc/memory/ , si evince che l'uso di StaticJsonBuffer è stato pensato proprio per evitare problemi di frammentazione. Per quello che ho capito, alloca, usa e successivamente libera la memoria non appena il buffer va fuori scope, proprio come una qualsiasi variabile.

closed 2019-09-18 by pat1