r-map/rmap #221: Flash firmware from SD Card

pat1, 2017-07-25:
L'uso di  https://github.com/JadinAndrews/avr_boot pare inibire il classico flash del firmware da seriale
come segnalato in https://github.com/JadinAndrews/avr_boot#serial-support---help-wanted

Non avendo questo loader controllo CRC può questo essere un problema complicando le fasi di diagnostica in caso di malfunzionamenti?

pat1, 2017-07-25:
Questa pull request potrebbe interessare:
https://github.com/zevero/avr_boot/pull/24

mbaldinetti, 2017-07-28:
il bootloader è stato fatto per funzionare anche sul 328p che ha un limite di dimensione del bootloader pari a 4Kb. Purtroppo la dimensione con fat32 è di poco inferiore ai 4Kb. L'autore del bootloader indica che è possibile includere il flash tramite seriale nella restante parte libera ma attualmente non è implementata.

in ogni caso, sul  644pa e sul 1284p, con una modifica dei fuse bit (molto utile il sito http://www.engbedded.com/fusecalc) è possibile aumentare la dimensione a 8Kb.

La mancanza del CRC non dovrebbe essere un problema (https://github.com/JadinAndrews/avr_boot#kiss). da quello che è indicato, basta effettuare un controllo CRC a posteriori, dopo aver creato il file bin.

mbaldinetti, 2019-10-03:
Attraverso digitecoboot è possibile eseguire il flash del firmware da sdcard o da porta seriale

closed 2019-10-03 by mbaldinetti