r-map/rmap #239: 6.2.3.4	Dimensionamento del pannello fotovoltaico e della batteria

pat1, 2017-10-17:
Dimensionare pannello fotovoltaico e batteria secondo la valutazione dei consumi della stazione e dei seguenti parametri:
garantire il funzionamento della stazione per almeno 7 giorni, anche in assenza totale di insolazione
la completa ricarica delle batterie in 18 ore alla potenza di picco del pannello.

pat1, 2017-12-11:
Per chiudere questa issue mi occorrno i dati tecnici della batteria e del pannello fotovoltaico.
Visti i tempi molto stretti risulta difficile fare una prova reale per valutare i consumi della stazione: serve quindi almeno una stima dei consumi (in una ora) della stazione.

mbaldinetti, 2017-12-12:
- Il pannello è da 20 W e fornisce (a potenza massima) 1.16 A ad una tensione di 17.3 V.
- La batteria è del tipo 12V e 18 Ah .
- La stazione nella configurazione MASTER GSM + I2C-TH + I2C-RAIN con caricabatterie ha un consumo medio (misurato) di 40 mAh a 12 V.

- Calcolo del massimo tempo di funzionamento in assenza di alimentazione:
18000 mAh / 40 mAh = 450 ore ~ 18 giorni di funzionamento (teorici, corrispondenti alla scarica completa della batteria).
Ipotizzando di lasciar carica la batteria al 50%, i giorni di funzionamento garantiti sono 9.

- Verifica del tempo di carica in 18 ore della batteria alla potenza di picco del pannello:
18 Ah / 1.16 A ~ 15 ore per la carica completa

closed 2017-12-19 by pat1