r-map/rmap #276: porting to python3: Exception Value: unorderable types: str() < NoneType()

pat1, 2018-08-29:
Quering borinud for summaries as:
```
 http://rmap.it/borinud/api/v1/dbajson/digiteco/1209812,4483519/fixed/254,0,0/103,2000,-,-/B12101/summaries
```

in MergeDB I get this error:
```
Environment:

Request Method: GET
Request URL: http://rmap.it/borinud/api/v1/dbajson/digiteco/1209812,4483519/fixed/254,0,0/103,2000,-,-/B12101/summaries

Django Version: 1.11.13
Python Version: 3.4.8
Installed Applications:
['django.contrib.auth',
 'django.contrib.contenttypes',
 'django.contrib.sessions',
 'django.contrib.sites',
 'django.contrib.messages',
 'django.contrib.admin',
 'django.contrib.staticfiles',
 'rmap.doc',
 'rmap',
 'rmap.stations',
 'rmap.network',
 'registration',
 'django_hosts',
 'leaflet',
 'djgeojson',
 'geoimage',
 'insertdata',
 'imagekit',
 'showdata',
 'amatyr',
 'borinud',
 'http2mqtt',
 'cookielaw',
 'graphite-dballe',
 'graphite-dballe.metrics',
 'graphite-dballe.render',
 'graphite-dballe.browser',
 'graphite-dballe.composer',
 'graphite-dballe.account',
 'graphite-dballe.dashboard',
 'graphite-dballe.whitelist',
 'graphite-dballe.events',
 'graphite-dballe.url_shortener',
 'tagging',
 'rainbo',
 'borinud_sos',
 'contacts',
 'firmware_updater']
Installed Middleware:
['django_hosts.middleware.HostsRequestMiddleware',
 'django.middleware.cache.UpdateCacheMiddleware',
 'django.contrib.sessions.middleware.SessionMiddleware',
 'django.middleware.locale.LocaleMiddleware',
 'django.middleware.common.CommonMiddleware',
 'rmap.stations.middleware.ProfileMiddleware',
 'django.middleware.csrf.CsrfViewMiddleware',
 'django.contrib.auth.middleware.AuthenticationMiddleware',
 'django.contrib.messages.middleware.MessageMiddleware',
 'django.middleware.cache.FetchFromCacheMiddleware',
 'django_hosts.middleware.HostsResponseMiddleware']

Traceback:

File "/usr/lib/python3.4/site-packages/django/core/handlers/exception.py" in inner
  41.             response = get_response(request)

File "/usr/lib/python3.4/site-packages/django/core/handlers/base.py" in _legacy_get_response
  249.             response = self._get_response(request)

File "/usr/lib/python3.4/site-packages/django/core/handlers/base.py" in _get_response
  187.                 response = self.process_exception_by_middleware(e, request)

File "/usr/lib/python3.4/site-packages/django/core/handlers/base.py" in _get_response
  185.                 response = wrapped_callback(request, *callback_args, **callback_kwargs)

File "/usr/lib/python3.4/site-packages/borinud/v1/views.py" in summaries
  189.         return JsonResponse(next(itertools.islice(dbajson(q,summary=True,format=format,dsn=request.GET.get('dsn', 'report'),seg=request.GET.get('seg', seg)),0,None)),safe=False)

File "/usr/lib/python3.4/site-packages/borinud/v1/views.py" in __iter__
  50.             self.handle = get_db(dsn=self.dsn,last=self.last).query_summary(self.q)

File "/usr/lib/python3.4/site-packages/borinud/utils/source.py" in query_summary
  287.             self.get_cached_summary()

File "/usr/lib/python3.4/site-packages/borinud/utils/source.py" in get_cached_summary
  230.             summary = self.set_cached_summary()

File "/usr/lib/python3.4/site-packages/borinud/utils/source.py" in set_cached_summary
  221.         } for o in res]

File "/usr/lib/python3.4/site-packages/borinud/utils/source.py" in <listcomp>
  212.         summary = [{

File "/usr/lib/python3.4/site-packages/borinud/utils/source.py" in query_summary
  155.             "query_summary", rec, reducer

File "/usr/lib/python3.4/site-packages/borinud/utils/source.py" in get_unique_records
  134.         ], key=self.unique_record_key), self.unique_record_key):

Exception Type: TypeError at /borinud/api/v1/dbajson/digiteco/1209812,4483519/fixed/254,0,0/103,2000,-,-/B12101/summaries
Exception Value: unorderable types: str() < NoneType()
```

edigiacomo, 2018-08-29:
The problem is the value used for the ordering comparisons, i.e. the tuple returned by `unique_record_keys`: some values can be null and the comparison with non-null values raises the error.

The error can be fixed by changing the value returned by `unique_record_keys`, e.g. we could convert the tuple `(ident, lon, lat, rep_memo, var, level, trange)` in a string (e.g. `digiteco/1209812,4483519/fixed/254,0,0/103,2000,-,-/B12101/`).

I'll create a PR as soon as possible.

edigiacomo, 2018-08-29:
I have pushed a fix in `python3` branch (dd296123c7c3424b3cadd6817b684170eda60c3f), please review it.

closed 2018-08-30 by pat1