r-map/rmap #277: porting to python3: Can't pickle <class 'Trange'>: attribute lookup Trange on builtins failed

pat1, 2018-08-30:
In borinud using cache
```
http://rmap.it/borinud/api/v1/dbajson/-/1129056,4462775/agrmet/254,0,0/103,2000,-,-/B12101/summaries
```
```
Environment:

Request Method: GET
Request URL: http://rmap.it/borinud/api/v1/dbajson/-/1129056,4462775/agrmet/254,0,0/103,2000,-,-/B12101/summaries

Django Version: 1.11.13
Python Version: 3.4.8
Installed Applications:
['django.contrib.auth',
 'django.contrib.contenttypes',
 'django.contrib.sessions',
 'django.contrib.sites',
 'django.contrib.messages',
 'django.contrib.admin',
 'django.contrib.staticfiles',
 'rmap.doc',
 'rmap',
 'rmap.stations',
 'rmap.network',
 'registration',
 'django_hosts',
 'leaflet',
 'djgeojson',
 'geoimage',
 'insertdata',
 'imagekit',
 'showdata',
 'amatyr',
 'borinud',
 'http2mqtt',
 'cookielaw',
 'graphite-dballe',
 'graphite-dballe.metrics',
 'graphite-dballe.render',
 'graphite-dballe.browser',
 'graphite-dballe.composer',
 'graphite-dballe.account',
 'graphite-dballe.dashboard',
 'graphite-dballe.whitelist',
 'graphite-dballe.events',
 'graphite-dballe.url_shortener',
 'tagging',
 'rainbo',
 'borinud_sos',
 'contacts',
 'firmware_updater']
Installed Middleware:
['django_hosts.middleware.HostsRequestMiddleware',
 'django.middleware.cache.UpdateCacheMiddleware',
 'django.contrib.sessions.middleware.SessionMiddleware',
 'django.middleware.locale.LocaleMiddleware',
 'django.middleware.common.CommonMiddleware',
 'rmap.stations.middleware.ProfileMiddleware',
 'django.middleware.csrf.CsrfViewMiddleware',
 'django.contrib.auth.middleware.AuthenticationMiddleware',
 'django.contrib.messages.middleware.MessageMiddleware',
 'django.middleware.cache.FetchFromCacheMiddleware',
 'django_hosts.middleware.HostsResponseMiddleware']

Traceback:

File "/usr/lib/python3.4/site-packages/django/core/handlers/exception.py" in inner
  41.             response = get_response(request)

File "/usr/lib/python3.4/site-packages/django/core/handlers/base.py" in _legacy_get_response
  249.             response = self._get_response(request)

File "/usr/lib/python3.4/site-packages/django/core/handlers/base.py" in _get_response
  187.                 response = self.process_exception_by_middleware(e, request)

File "/usr/lib/python3.4/site-packages/django/core/handlers/base.py" in _get_response
  185.                 response = wrapped_callback(request, *callback_args, **callback_kwargs)

File "/usr/lib/python3.4/site-packages/borinud/v1/views.py" in summaries
  189.         return JsonResponse(next(itertools.islice(dbajson(q,summary=True,format=format,dsn=request.GET.get('dsn', 'report'),seg=request.GET.get('seg', seg)),0,None)),safe=False)

File "/usr/lib/python3.4/site-packages/borinud/v1/views.py" in __iter__
  50.             self.handle = get_db(dsn=self.dsn,last=self.last).query_summary(self.q)

File "/usr/lib/python3.4/site-packages/borinud/utils/source.py" in query_summary
  310.             self.get_cached_summary()

File "/usr/lib/python3.4/site-packages/borinud/utils/source.py" in get_cached_summary
  253.             summary = self.set_cached_summary()

File "/usr/lib/python3.4/site-packages/borinud/utils/source.py" in set_cached_summary
  245.         self.cache.set('borinud-summary-cache-%s' % self.dsn, summary, self.timeout)

File "/usr/lib/python3.4/site-packages/django/core/cache/backends/filebased.py" in set
  57.                 f.write(zlib.compress(pickle.dumps(value, pickle.HIGHEST_PROTOCOL)))

Exception Type: PicklingError at /borinud/api/v1/dbajson/-/1129056,4462775/agrmet/254,0,0/103,2000,-,-/B12101/summaries
Exception Value: Can't pickle <class 'Trange'>: attribute lookup Trange on builtins failed
```

edigiacomo, 2018-09-04:
Please review commit 72b78e6.

pat1, 2018-09-05:
it works

closed 2018-09-05 by pat1