r-map/rmap #291: server: 'NoneType' object is not subscriptable in show all stations

pat1, 2019-02-22:
http://rmap.cc/borinud/api/v1/dbajson/-/*/*/*/*/*/stations?dsn=report

Traceback:

File "/usr/lib/python3.4/site-packages/django/core/handlers/exception.py" in inner
  41.             response = get_response(request)

File "/usr/lib/python3.4/site-packages/django/core/handlers/base.py" in _legacy_get_response
  249.             response = self._get_response(request)

File "/usr/lib/python3.4/site-packages/django/core/handlers/base.py" in _get_response
  187.                 response = self.process_exception_by_middleware(e, request)

File "/usr/lib/python3.4/site-packages/django/core/handlers/base.py" in _get_response
  185.                 response = wrapped_callback(request, *callback_args, **callback_kwargs)

File "/usr/lib/python3.4/site-packages/borinud/v1/views.py" in stations
  295.         return JsonResponse(next(itertools.islice(dbajson(q,stations=True,format=format,dsn=request.GET.get('dsn', 'report'),seg=request.GET.get('seg', 'hostorical')),0,None)),safe=False)

File "/usr/lib/python3.4/site-packages/borinud/v1/views.py" in __next__
  66.         for self.s in self.handle:

File "/usr/lib/python3.4/site-packages/borinud/utils/source.py" in query_stations
  162.             "query_stations", rec, lambda g: next(g)

File "/usr/lib/python3.4/site-packages/borinud/utils/source.py" in get_unique_records
  157.         ], key=self.unique_record_key), self.unique_record_key):

File "/usr/lib/python3.4/site-packages/borinud/utils/source.py" in unique_record_key
  144.             rec.get("trange")[0],

Exception Type: TypeError at /borinud/api/v1/dbajson/-/*/*/*/*/*/stations
Exception Value: 'NoneType' object is not subscriptable

edigiacomo, 2019-12-31:
Does this problem still persist?

pat1, 2019-12-31:
exist in old version, not supported now

closed 2019-12-31 by pat1