r-map/rmap #298: firmware: stima v3 do not compile

pat1, 2019-09-11:
`
/autofs/nfshomes/ppatruno/git/rmap/arduino/sketchbook/stima_v3/stima/stima.ino: In function 'void supervisor_task()':
stima:1227: error: 'SDCARD_TEST_FILE_NAME' was not declared in this scope; did you mean 'SDCARD_MQTT_PTR_FILE_NAME'?
 1227 |             if (sdcard_open_file(&SD, &test_file, SDCARD_TEST_FILE_NAME, O_RDWR | O_CREAT | O_APPEND)) {
      |                                                   ^~~~~~~~~~~~~~~~~~~~~
      |                                                   SDCARD_MQTT_PTR_FILE_NAME
/autofs/nfshomes/ppatruno/git/rmap/arduino/sketchbook/stima_v3/stima/stima.ino: In function 'void time_task()':
stima:1383: error: 'NTP_MAX_DIFF_VALID_TIME_S' was not declared in this scope
 1383 |          if ((current_ntp_time > NTP_VALID_START_TIME_S) && (diff_ntp_time <= NTP_MAX_DIFF_VALID_TIME_S)) {
      |                                                                               ^~~~~~~~~~~~~~~~~~~~~~~~~
`

It's correct SDCARD_TEST_FILE_NAME is not defined:

`
grep -r SDCARD_TEST_FILE_NAME *
stima_v3/stima/stima.ino:            if (sdcard_open_file(&SD, &test_file, SDCARD_TEST_FILE_NAME, O_RDWR | O_CREAT | O_APPEND)) {
`

pat1, 2019-09-11:
closed by  1d58c25

pat1, 2019-09-11:
`
/autofs/nfshomes/ppatruno/git/rmap/arduino/sketchbook/libraries/SensorDriver/SensorDriver.cpp: In member function 'virtual void SensorDriverOpc::get(int32_t*, uint8_t)':
/autofs/nfshomes/ppatruno/git/rmap/arduino/sketchbook/libraries/SensorDriver/SensorDriver.cpp:2201:9: error: 'bin' was not declared in this scope
         bin = UINT16_MAX;
         ^
/autofs/nfshomes/ppatruno/git/rmap/arduino/sketchbook/libraries/SensorDriver/SensorDriver.cpp:2205:9: error: 'bin' was not declared in this scope
         bin = UINT16_MAX;
         ^
/autofs/nfshomes/ppatruno/git/rmap/arduino/sketchbook/libraries/SensorDriver/SensorDriver.cpp:2209:9: error: 'val' was not declared in this scope
         val = UINT16_MAX;
         ^
`

mbaldinetti, 2019-10-03:
closed by 2ea353342e1460453b20c646eb4f5eb288d11e86

closed 2019-10-03 by mbaldinetti