r-map/rmap #307: server: arkimet scan for generic mobile messages

pat1, 2020-02-12:
on old version of arkimet with lua was:
  function scan(msg, md)
    local proddef = bufr_read_proddef(msg)
    local report = msg:find("rep_memo")
    local area = nil
    if report:enqc() == "mobile"
    then
       area = bufr_read_area_mobile(msg)
    else
       area = bufr_read_area_fixed(msg)
    end
    if area then md:set(arki_area.grib(area)) end
    if proddef then md:set(arki_proddef.grib(proddef)) end
    if report then md:set(md.product:addValues{t=report:enqc()}) end
    md:set(bufr_scan_forecast(msg))
  end

this needs porting to new arkimet with python

pat1, 2020-02-12:
read at https://github.com/ARPA-SIMC/arkimet/issues/199

edigiacomo, 2020-02-12:
Can you give me a BUFR file with a "mobile" report?

pat1, 2020-02-19:
eccolo:
[01.bufr.zip](https://github.com/r-map/rmap/files/4224854/01.bufr.zip)

edigiacomo, 2020-02-19:
Create file `/etc/arkimet/scan/bufr_generic_mobile_rmap.py`:

```python
from arkimet.scan.bufr import Scanner, read_area_mobile, read_proddef

def scan(msg, md):
    if msg.report == "mobile":
        area = read_area_mobile(msg)
        proddef = read_proddef(msg)
        if area:
            md["area"] = {"style": "GRIB", "value": area}
        if proddef:
            md["proddef"] = {"style": "GRIB", "value": proddef}
    else:
        return False

Scanner.register("generic", scan, priority=1)
```

closed 2020-03-10 by pat1