r-map/rmap #329: Missing negative temperature data

mbaldinetti, 2021-02-04:
Missing negative temperature data

pat1, 2021-02-05:
avevo richiesto una pull request ...
riapro per fare una revert , tag pull request etc

pat1, 2021-02-10:
f01cbab siamo in due in ARPAE a non comprendere il senso della patch
le due variabili sono float e fare la differenza dei loro abs o l'abs della differenza dovrebbe essere abbastanza indifferente salvo forse situazioni di underflow e overflow
@mbaldinetti sapresti spiegarci il senso della patch ?

pat1, 2021-02-15:
anzi l'ordine di esecuzione degli abs non è indifferente nel caso di temperature misurata e di conferma con segno differente, ma risulterebbe corretta la formula prima della patch e non dopo

mbaldinetti, 2021-02-19:
dopo alcune verifiche confermo che la patch è errata.
Il codice era già correttamente funzionante così senza bug delle temperature negative.

commit f01cbab4 (2021-02-04): fix #329
commit a357421c (2021-02-19): Revert "fix #329"

This reverts commit f01cbab460e74edfc16b1308ce1d5c4943729ab0.
closed 2021-02-19 by mbaldinetti