r-map/rmap #356: server: mgrate from the use of ident for authorization to user + station/ident

pat1, 2021-11-20:
Autorizzazione e tracciabilità attualmente usano in modo improprio ident.
E' necessario migrare a un sistema basato su user + station (ident+latlon+network) per le stazioni fisse e user + ident per le stazioni mobili.

* MQTT: modifica del topic di pubblicazione 
  - Stato della connessione
    maint/USER/IDENT/COORDS/NETWORK/254,0,0/265,0,-,-/B01213/
  - Data e
    <rootpath>/USER/IDENT/COORDS/NETWORK/TRANGE/LEVEL/VAR
  - Constant Data
    <rootpath>/USER/IDENT/COORDS/NETWORK/-,-,-/-,-,-,-/VAR
  - RPC
    topiccom="rpc/USER/IDENT/COORDS/NETWORK/com"
    topicres=  "rpc/USER/IDENT/COORDS/NETWORK/res"

Il <rootpath> fixed/mobile attualmente determina la destinazione nei DB ma potrebbe essere utilizzata la presenza o no di IDENT

* server cambio autorizzazione
  - senza ident (stazioni fisse) verifica esistenza della stazione per quell'utentre
  - con ident (stazioni mobili) verifica esistenza ident registrato per quell'utente

* server cambio tracciabilità
  A ogni dato accolto dovranno essere addizionati alcuni metadati nella sezione constant station data corrispondenti a:
  - user
  - altri dati di tracciabilità del dato

* gestione dati tracciabilità: in tutto il percorso dei dati dovrà essere garantita la persistenza e coerenza dei dati di tracciabilità nei constant station data: a tal proposito dballe dovrà garantire una corretta gestione di tali metadati

* sistema di autorizzazione: è possibile ipotizzare una autenticazione per stazione e non per utente: username/password per utente e differenti username/password per ogni stazione gestiti dall'utente

pat1, 2022-05-02:
Questo approccio ha un problema: ogni utente può decidere liberamente gli ident da utilizzare per stazioni mobili il che comporta nel data model RMAP possibili e inevitabili conflitti, ossia differenti utenti potrebbero utilizzare lo stesso ident essendo lo user non facente parte del datamodel dei metadati di identificazione univoca del dato.

pat1, 2022-05-02:
Per ora è stato modificato il sistema di autenticazione:
ogni utente gestisce le sue stazioni composte da una o più board, ogni board con più trasporti, ma solo uno per tipo.
Il trasporto mqtt definisce il server, utente, password e pskkey
l'autenticazione di una stazione avviene tramite:
user/station_slug/board_slug
password oppure pskkey

quindi ogni board di una stazione si autentica con credenziali differenti e viene autorizzata sui suoi specifici topics; ad esempio:
```
user stima37
pattern sample/stima37/1146210,4440275/fixed/
pattern report/stima37/1146210,4440275/fixed/
pattern maint/stima37/1146210,4440275/fixed/
pattern rpc/stima37/1146210,4440275/fixed/
```
ossia 4 topics per ogni stazione fissa

l'implementazione non è risolta per le stazioni mobili in quanto per il momento per non creare conflitti di ident rimane possibile registrare una sola stazione mobile che ha per ident lo user corrispondente con i seguenti topics:
```
user rmap
pattern sample/rmap/+/mobile/
pattern report/rmap/+/mobile/
pattern maint/rmap/+/mobile/
pattern rpc/rmap/+/mobile/
```

edigiacomo, 2022-05-03:
Da riunione odierna: la soluzione temporanea è di avere una stazione mobile per utente con ident pari a nome utente. In futuro si potrà prevedere una associazione tra utente e più ident univoci che comporta dello sviluppo.

pat1, 2025-05-20:
la condizione di avere una stazione mobile per utente con ident pari a nome utente è stata imposta nel metodo clean nel modello dati di Django oltre alla verifica dell'aderenza di ident alla codifica bufr WMO b5832f23bfc5ae82a179029877565a4b6ed66f95