r-map/rmap #358: Firmware: I2C lcd display stop working after sometime

PasqualiSIMC, 2021-11-24:
Il display di diverse stazioni stima della rete claster risulta con i pixel neri dopo alcuni mesi di funzionamento. Lo scollegamento dal modulo e il reset della stazione non risolvono il problema

pat1, 2021-11-24:
non ho al momento a disposizione un display, ma bisognerebbe controllare come avviene la regolazione di contrasto

pat1, 2021-11-26:
[image]

da provare una regolazione di contrasto tramite il trimmer indicato

pat1, 2021-11-26:
L'eventuale restart del display con:
LCD_BEGIN(&lcd, LCD_COLUMNS, LCD_ROWS);

ha queste tempistiche: 1100 millisecondi

pat1, 2021-11-26:
Disabilitando queste linee nella libreria LiquidCristal_i2c
```
	/*
	// SEE PAGE 45/46 FOR INITIALIZATION SPECIFICATION!
	// according to datasheet, we need at least 40ms after power rises above 2.7V
	// before sending commands. Arduino can turn on way befer 4.5V so we'll wait 50
	delay(50); 
  
	// Now we pull both RS and R/W low to begin commands
	expanderWrite(_backlightval);	// reset expanderand turn backlight off (Bit 8 =1)
	delay(1000);
	*/
```
il begin del display impiega 51 millisecondi; che mi pare compatibile con i tempi "non blocking"

pat1, 2021-11-27:
credo sia molto meglio migrare a hd44780 library:
https://github.com/duinoWitchery/hd44780

commit 386d7d15 (2021-11-29): close #363 close #362 close #358