r-map/rmap #398: hardware: I2C HUB step down Buck Converter is very hot without load

pat1, 2022-01-28:
Even without load (no cable connected and no GS connected) the TLV62569 in the I2C hub module is very hot.

Looking at schematic the C14 (100uF) is not conform to data-sheet https://www.ti.com/lit/ds/symlink/tlv62569.pdf in table 4 the recommended value is 10 uF.

pat1, 2022-02-01:
[image]

un TLV62569 si è probabilmente bruciato, ma comunque con una board perfettamente funzionante il ripple a 3.3V è comunque di 0.3V con 3 chip e relative resistenze di pull up abilitati e null'altro collegato

mbaldinetti, 2022-02-02:
i ripple vanno sempre ottimizzati in base alla corrente di carico presunta che immagino sia intorno a 0.5/1A. In ogni caso non ho seguito personalmente la progettazione del circuito switching, ma non abbiamo mai avuto problemi con l'hardware ad esso collegato.

pat1, 2022-02-02:
tralasciando il ripple avanza la questione del condensatore 10 volte il valore consigliato:
Looking at schematic the C14 (100uF) is not conform to data-sheet https://www.ti.com/lit/ds/symlink/tlv62569.pdf in table 4 the recommended value is 10 uF.