r-map/rmap #405: firmware: reboot i2c at first error

mbaldinetti, 2022-02-03:
restarting the i2c bus at each error makes the system more responsive in case of multiple errors on the bus also due to other modules

pat1, 2022-02-03:
non sono convinto di questa strategia in senso generale, il bus I2C deve funzionare ed essere stabile prima di tutto.
Il reset porta a una temporanea indisponibilità a rispondere a richieste tipo request che anche se non utilizzata non può comunque essere implementeta per uso generale.
Con l'attuale funzionamento sul modulo master in corso di test non è MAI stato necessario un reset del bus.
Inoltre sul modulo master (unico modulo) il reset del bus è un HARD reset e porta down per un certo tempo SDA e SCL quindi è distruttivo per il bus.
Si può applicare questa strategia per gli altri modulo ma NON per il modulo master.

pat1, 2022-02-03:
guardare a 
https://github.com/r-map/rmap/blob/0b20edf5572831bc5069464f7f5a1e8b7f4da1ec/platformio/stima_v3/stima/src/stima.ino#L378
operazione necessaria per recuperare da blocchi ben documentati su I2C

mbaldinetti, 2022-02-03:
è ovvio che è una situazione al limite ma con tanti disturbi esterni sul bus, per ora, è l'unico modo che lo rende funzionante. se si trovano strategie migliori, ben vengano

pat1, 2022-02-15:
tutti i test superati con successo compreso con il generatore di rumore i2c_noise_generator che simula una situazione completamente degenerata.

commit 259c71cc (2022-02-03): close #405
closed 2022-02-15 by pat1