r-map/rmap #406: firmware: for very big number MQTT_MESSAGE_LENGTH is too small

mbaldinetti, 2022-02-03:
for very big number MQTT_MESSAGE_LENGTH is too small

pat1, 2022-02-03:
ogni volta che si modifica questo valore bisogna ricordarsi di cambiare anche il programma che legge i dati da SDcard:
https://github.com/r-map/rmap/blob/0b20edf5572831bc5069464f7f5a1e8b7f4da1ec/mqtt2bufr/mqtt2bufr.py#L219

consiglierei di fare una attenta stima anche per la trasmissione delle tabelle D per qualità aria e vento e poi mantenerlo

pat1, 2022-02-04:
per il topic:
```
254,0,3600/103,20000,103,20000/B12101
```
lunghezza max di 37 caratteri 

il payload più lungo in tabella B senza attributi risulta essere:
```
{"v":-2147483648,"t":"2022-02-04T12:45:00"}
```
di 43 caratteri
quindi
MQTT_SENSOR_TOPIC_LENGTH = 37
MQTT_SENSOR_TOPIC_LENGTH = 43+1 = 44

nel caso di tabella d vento:
{"d":51,"p":[-2147483648,-2147483648,-2147483648,-2147483648,-2147483648,-2147483648],"t":"2022-02-04T13:15:00"}
nel caso tabella d air quality:
```
{"d":52,"p":[-2147483648,-2147483648,-2147483648,-2147483648,-2147483648,-2147483648,-2147483648,-2147483648,-2147483648,-2147483648,-2147483648,-2147483648,-2147483648],"t":"2022-02-04T13:00:00"}
```
MQTT_SENSOR_TOPIC_LENGTH = 195+1 = 196

Valuterei la possibilità di scrivere queste due lunghezze in un file (test.txt) sulla SDcard in modo da poterle configurare a piacere

pat1, 2022-02-04:
ho inserito la scrittura di una serie di informazioni utili comprese MQTT_MESSAGE_LENGTH MQTT_SENSOR_TOPIC_LENGTH e il root topic in un file su sd card poi letto da mqtt2bufr dalla versione 3.4
committato nel branch master con le nuove lunghezze

closed 2022-02-15 by pat1