r-map/rmap #436: stimav4: master board do not compile

pat1, 2023-02-23:
```
[pat1@localhost master]$ pio run
Processing stimav4_master (board: stimav4_master; platform: ststm32; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Error: Unknown board ID 'stimav4_master'
```

digitecomg, 2023-02-23:
All'interno della cartella stima4v c'è una cartella boards
i files json vanno caricati nelle boards arduino:
stimav4_master.json e stimav4_slave.json in .platformio\platforms\ststm32\boards

Le directory (interessano poi solo i files variant_stima*.*)
 - L452R(C-E)(I-T-Y)_L462RE(I-T-Y)
 - L496V(E-G)T_L4A6VGT
Nelle directory dei variant in .platformio\packages\framework-arduinoststm32\variants\STM32L4xx

Successivamente seguire una nota all'interno del file note nella cartella user_custom di master

pat1, 2023-04-14:
ported configuration to platformio.ini

closed 2023-04-14 by pat1