r-map/rmap #441: stimav4: modulo svwc soil volumetric water content implementation

pat1, 2023-06-07:
ToDo: Implement and test the module

digitecomg, 2023-07-14:
Il modulo è stato inserito ed attualmente pubblica i dati in questo modo:

>1/maint/userv4//1212345,4512345/test/254,0,0/265,0,-,-/B01213 {"t":"2023-07-14T18:24:00", "bs":"module_svwc", "b":"0b0000000000000000", "c":[0,0,0,0]} [ OK ]

>1/report/userv4//1212345,4512345/test/0,0,180/106,250,-,-/B13227 {"v":38,"t":"2023-07-14T18:24:00","a":{"B33199":100}} [ OK ]
>1/report/userv4//1212345,4512345/test/0,0,180/106,450,-,-/B13227 {"v":38,"t":"2023-07-14T18:24:00","a":{"B33199":100}} [ OK ]
>1/report/userv4//1212345,4512345/test/0,0,180/106,700,-,-/B13227 {"v":38,"t":"2023-07-14T18:24:00","a":{"B33199":100}} [ OK ]

Nel server probabilmente non è stata ancora implementata la visualizzazione

pat1, 2023-07-15:
Configurazione della stazione e visualizzazione dei dati sono sincrone.
Se la stazione è configurata per avere un modulo svwc i dati dovrebbero essere visualizzati.
Quale stazione è utilizzata per i test ?

digitecomg, 2023-07-15:
La solita
https://test.rmap.cc/stationconfig/userv4/verifica
Dove ho anche abilitato la pioggia con le 24 ore.

{"jsonrpc": "2.0", "method": "configure", "params": {"board": "module_svwc"}}
{"jsonrpc": "2.0", "method": "configure", "params": {"boardtype": 29, "sn": "0xabcdef6"}}
{"jsonrpc": "2.0", "method": "configure", "params": {"reset": true}}
{"jsonrpc": "2.0", "method": "configure", "params": {"cansampletime": 60}}
{"jsonrpc": "2.0", "method": "configure", "params": {"node_id": 65, "subject": "node.svwc", "subject_id": 65}}
{"jsonrpc": "2.0", "method": "configure", "params": {"driver": "CAN", "type": "SVW", "timerange": [0, 0, 180], "level": [106, 250, null, null]}}
{"jsonrpc": "2.0", "method": "configure", "params": {"driver": "CAN", "type": "SVW", "timerange": [0, 0, 180], "level": [106, 450, null, null]}}
{"jsonrpc": "2.0", "method": "configure", "params": {"driver": "CAN", "type": "SVW", "timerange": [0, 0, 180], "level": [106, 700, null, null]}}
{"jsonrpc": "2.0", "method": "configure", "params": {"save": true}}

pat1, 2023-07-15:
se qualche cosa relativa a una issue continua a non funzionare non si può chiudere la issue.
Si può chiudere e aprirne un'altra, oppure assegnarla a qualcun altro.
I dati sul server continuano a non vedersi.

pat1, 2023-07-15:
Anche le board disattivate venivano configurate.
Attivata la board i dati sono comparsi sul server.

related to fd899aa55

closed 2023-07-15 by pat1