r-map/rmap #442: stimav4: module rain, some measures are missed

pat1, 2023-06-07:
Nel modulo precipitazione alcune misure risultano mancanti:

il calcolo delle intensità di precipitazione dovrà seguire il seguente procedimento

Elaborazione dei campionamenti per passare da level I a level II :
● calcolo delle precipitazioni a intervalli di 60”
● calcolo del valore massimo delle precipitazioni a intervalli di 60”
● calcolo del valore massimo dei valori di precipitazione a intervalli di 60” con finestra mobile di 300” a passi di 60”

Elaborazioni statistiche utili per la composizione di un report:
● cumulata nell’intervallo del report
● valore massimo nel periodo dell’intensità di precipitazione in 60 secondi
● valore massimo dell’intensità di precipitazione in 300 secondi

è stato aggiunto quindi un sensore tipo "TPR" e due nuove variabili:
*  013208 Total precipitation rate over 1'                                 Kg/m**2/s
* 013209 Total precipitation rate over 5'                                 Kg/m**2/s

le variabili sono rappresentabili con un minimo di 20 bit e con la precisione a un decimale

pat1, 2023-06-07:
Rispetto a quanto proposto come algoritmo e discusso a voce dopo ampia discussione interna propopngo quanto segue ipotizzando un report ogni 15':

B13208 : calcolo delle precipitazioni a intervalli di 60” (15 valori); B13208 corrisponde al valore massimo dei precedenti 15 valori  (nessun cambiamento a quanto scritto precedentemente)

B13209 : calcolo delle precipitazioni a intervalli di 5' (3 valori); B13209 corrisponde al valore massimo dei precedenti 3 valori  (scritto precedentemente era un valore massimo mobile sui 15 valori calcolati per B13208)

pat1, 2023-06-07:
Qui alla stazione userv4 verifica è già disponibile una configurazione completa:

https://test.rmap.cc/stationconfig/userv4/verifica

digitecomg, 2023-07-11:
B13208 : calcolo delle precipitazioni a intervalli di 60” (15 valori); B13208 corrisponde al valore massimo dei precedenti 15 valori (nessun cambiamento a quanto scritto precedentemente)
Dividiamo in 15 acquisizione precise e calcoliamo il massimo tra 15 valori o è il massimo valore di 60 secondi consecutivi all'interno dei 15 minuti? Mi sembrava di aver capito la seconda ipotesi.

B13209 : calcolo delle precipitazioni a intervalli di 5' (3 valori); B13209 corrisponde al valore massimo dei precedenti 3 valori (scritto precedentemente era un valore massimo mobile sui 15 valori calcolati per B13208)
Quindi la cumulata massima in 5 minuti prendendo i riferimenti 0-5 / 5-10 / 10-15 (ovviamente il massimo valore)

digitecomg, 2023-07-11:
Qui alla stazione userv4 verifica è già disponibile una configurazione completa:

https://test.rmap.cc/stationconfig/userv4/verifica

Riesci a ripristinare la configurazione con TPR?

digitecomg, 2023-07-11:
le variabili sono rappresentabili con un 14 bit e con la precisione alla quarta cifra decimale (offset 0 e scale 10^4)
Se ci sono ad. es. 19,66 mm in 5 minuti (non impossibile) (0,065535) puoi usare 10^6 MAX con precisione massima direi alla seconda cifra decimale e l'utilizzo di tutti i 16 bit.

pat1, 2023-07-11:
Non ho capito.
ci sono problemi ad usare queste definizioni ?

    B13208 Total precipitation rate over 1' Kg/m**2/s
    B13209 Total precipitation rate over 5' Kg/m**2/s

le variabili sono rappresentabili con un 14 bit e con la precisione alla quarta cifra decimale (offset 0 e scale 10^4)

pat1, 2023-07-11:
Confermo che la proposta è di NON utilizzare finestre mobili, ma intervalli sincroni di 60" e 5'
 
> B13208 : calcolo delle precipitazioni a intervalli di 60” (15 valori); B13208 corrisponde al valore massimo dei precedenti 15 valori (nessun cambiamento a quanto scritto precedentemente)
Dividiamo in 15 acquisizione precise e calcoliamo il massimo tra 15 valori o è il massimo valore di 60 secondi consecutivi all'interno dei 15 minuti? Mi sembrava di aver capito la seconda ipotesi.

> B13209 : calcolo delle precipitazioni a intervalli di 5' (3 valori); B13209 corrisponde al valore massimo dei precedenti 3 valori (scritto precedentemente era un valore massimo mobile sui 15 valori calcolati per B13208)
Quindi la cumulata massima in 5 minuti prendendo i riferimenti 0-5 / 5-10 / 10-15 (ovviamente il massimo valore)

digitecomg, 2023-07-11:
Confermo che la proposta è di NON utilizzare finestre mobili, ma intervalli sincroni di 60" e 5'

Peccato... avevo fatto la finestra mobile nei 60". La tolgo

B13208 Total precipitation rate over 1' Kg/m**2/s
B13209 Total precipitation rate over 5' Kg/m**2/s

Per la rappresentazione possiamo usare 10^5 e in 14 bit ci stà un valore cumulato mediato di 49 mm in 5 minuti

digitecomg, 2023-07-11:
Ho inserito una define per utilizzare eventualmente la modalità mobile o fissa, così non l'ho eliminata.

pat1, 2023-07-12:
> Per la rappresentazione possiamo usare 10^5 e in 14 bit ci stà un valore cumulato mediato di 49 mm in 5 minuti

la risoluzione  utile è di 0.1Kg/m^2 in 5 minuti pari a 0,000333333 Kg/m^2/s
Quindi 10^4 dovrebbe essere sufficiente

con 14 bit si rappresentano rate di 16383*10^-4  ~ 1.6Kg/m^2/s pari a una pioggia di 98 Kg/m^2 in un minuto.
Non ci sono molti dati disponibili sul minuto ma piogge di  ~ 10 Kg/m in 5 minuti sono frequenti, sul minuto non possono essere altro che valori più elevati e quindi mi pare che l'ordine di grandezza sia corretto.
Probabile che i pluviometri a basculata inizino  ad avere seri problemi a basculare 1 volta al secondo ...

pat1, 2023-07-12:
http://www.bom.gov.au/water/designRainfalls/rainfallEvents/worldRecRainfall.shtml

visto che la definizione della variabile è indipendente dallo strumento utilizzato (possibile utilizzare anche il radar meteo)
lascerei offset e scale così come originariamente proposto.

pat1, 2023-07-12:
> Riesci a ripristinare la configurazione con TPR?

fatto

```
{"jsonrpc": "2.0", "method": "configure", "params": {"node_id": 61, "subject": "node.p", "subject_id": 61}}
{"jsonrpc": "2.0", "method": "configure", "params": {"driver": "CAN", "type": "TBR", "timerange": [1, 0, 180], "level": [1, null, null, null]}}
{"jsonrpc": "2.0", "method": "configure", "params": {"driver": "CAN", "type": "TPR", "timerange": [2, 0, 180], "level": [1, null, null, null]}}
```

digitecomg, 2023-07-13:
Attualmente spedisco questi dati ma non vedo riscontro sul server, già calcolati e collegati in automatico.
La stringa è corretta o c'è qualcosa che non va?

>1/report/userv4//1212345,4512345/test/1,0,180/1,-,-,-/B13011 {"v":77,"t":"2023-07-13T17:33:00","a":{"B33199":70}} [ OK ]
>1/report/userv4//1212345,4512345/test/2,0,180/1,-,-,-/B13208 {"v":4333,"t":"2023-07-13T17:33:00","a":{"B33199":70}} [ OK ]
>1/report/userv4//1212345,4512345/test/2,0,180/1,-,-,-/B13209 {"v":null,"t":"2023-07-13T17:33:00","a":{"B33199":70}} [ OK ]

pat1, 2023-07-14:
un problema di aggiornamento sul server ora risolto.

digitecomg, 2023-07-14:
Parametri aggiunti

closed 2023-07-14 by digitecomg