r-map/rmap #449: stimav4: module master do not compile

pat1, 2023-06-07:
```
In file included from src/canard_class_master.cpp:31:
include/canard_class_master.hpp:52:10: fatal error: rmap/_module/RAIN_1_0.h: No such file or directory
   52 | #include <rmap/_module/RAIN_1_0.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio/build/stimav4_master/src/canard_class_master.cpp.o] Error 1
========================================================================================== [FAILED] Took 69.31 seconds ==========================================================================================
```

pat1, 2023-11-02:
duplicated of #459

pat1, 2024-03-27:
from [Continuous integration](https://en.wikipedia.org/wiki/Continuous_integration)
``
/home/runner/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/12.3.1/../../../../arm-none-eabi/bin/ld: .pio/build/stimav4_master/lib685/CycloneTCP-DriversUart/drivers/uart/uart_driver.c.o: in function `USART2_IRQHandler':
/home/runner/work/rmap/rmap/platformio/stima_v4/libraries/CycloneTCP-DriversUart/src/drivers/uart/uart_driver.c:159: multiple definition of `USART2_IRQHandler'; .pio/build/stimav4_master/SrcWrapper/src/stm32/uart.c.o:/home/runner/.platformio/packages/framework-arduinoststm32/libraries/SrcWrapper/src/stm32/uart.c:976: first defined here
/home/runner/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/12.3.1/../../../../arm-none-eabi/bin/ld: warning: .pio/build/stimav4_master/firmware.elf has a LOAD segment with RWX permissions
collect2: error: ld returned 1 exit status
*** [.pio/build/stimav4_master/firmware.elf] Error 1
``

digitecomg, 2024-03-27:
Nel modulo -> alla linea 976
.pio/build/stimav4_master/SrcWrapper/src/stm32/uart.c.o:/home/runner/.platformio/packages/framework-
arduinoststm32/libraries/SrcWrapper/src/stm32/uart.c:976

#if defined(USART2_BASE)
__weak void USART2_IRQHandler(void)
{

VA Inserito __weak davanti alla definizione della funzione ->void USART2_IRQHandler(void)

Come già fatto in passato

closed 2024-05-22 by pat1