r-map/rmap #452: stimav4: wind measure

pat1, 2023-06-08:
il test è effettuato con la seguente configurazione del sensore fake:
```
// Gill, Polar, Polled
#define POLL_MODE true

// rotation period and velocity
#define ROTATION_MINUTES    10
#define ROTATION_VELOCITY   20.D

// constant u component
//#define CONSTANT_U_VELOCITY 5.D
#define CONSTANT_U_VELOCITY 10.D
// random u component
//#define RANDOM_U_VELOCITY   random(-1000,1001)*0.01D
#define RANDOM_U_VELOCITY   0.0D

// constant v component
//#define CONSTANT_V_VELOCITY 5.D
#define CONSTANT_V_VELOCITY 10.D
// random v component
//#define RANDOM_V_VELOCITY   random(-1000,1001)*0.01D
#define RANDOM_V_VELOCITY   0.0D
```
il report ha un periodo di 15'

### risultati

**vento intensità:**
teorico = sqrt(10*10+10*10) = 14.14
https://test.rmap.cc/graphite/S/Ex

**vento direzione**
teorico 225
https://test.rmap.cc/graphite/S/Ey

**media vettoriale direzione e intensità**
teorico: ogni report sfasamento di 5' corriposndente a un semiperiodo di rotazione del vento
quindi rotazione della media vettoriale ogni 2 report (30')
https://test.rmap.cc/graphite/S/FA
https://test.rmap.cc/graphite/S/FB

**max wind gust speed**
teorico = 20+sqrt(10*10+10*10) = 34.14
https://test.rmap.cc/graphite/S/FC**

**max wind gust direction**
**teorico 225
https://test.rmap.cc/graphite/S/FD**

**max wind gust speed log period**
teorico = poco meno di 20+sqrt(10*10+10*10) = 34.14
https://test.rmap.cc/graphite/S/FF

**max wind gust direction long period**
teorico intorno a 225
https://test.rmap.cc/graphite/S/FE

pat1, 2023-06-08:
l'unico problema evidenziato è relativo a max wind gust direction
che dovrebbe presentare un valore stabile teorico di 225 gradi

pat1, 2023-07-19:
rieffettuati i test si conferma il problema su max wind gust direction
che dovrebbe presentare un valore stabile teorico di 225 gradi e invece presenta un valore oscillante tra 221 e 222 gradi
https://test.rmap.cc/graphite/S/FL
Il simulatore fa un angolo giro in 10 minuti e viene campionato ogni secondo quindi con una risoluzione di 360/600
= 0.6 gradi quindi con la risoluzione  del windsonic dovremmeo avere un errore inferiore pari al grado. 
Possibile un errore di troncamento?

digitecomg, 2023-07-19:
Non penso il troncamento. Venerdì ricontrollo quel parametro calcolato con tutti i dati che generano il risultato sui 15' e capiremo l'origine del problema. Ti aggiorno appena fatto.

digitecomg, 2023-07-21:
Risolto il problema con aggiunta di un decimale nell'acquisizione. Confermato il problema di troncamento non nell'elaborazione ma nell'acquisizione dal sensore. Il dato con questo tipo di test è possibile al grado come previsto. Rieffetuata la prova con il simulatore dopo la verifica con il debugger e la serie dati ricevuta.