r-map/rmap #453: stimav4: temperature and humidity measure

pat1, 2023-06-08:
Il simulatore ha queste impostazioni
**temperatura**
	int baset = 27315;
	int rndt = random(-300,301);
	int dailyt = round(sin((float(millis()-starttime)/(60.*60.*24.*1000.))*2.*PI)*2000.);
	t = baset + rndt + dailyt;

**umidità**
	int baseh = 50;
	int rndh = random(-5,6);
	int dailyh = - round(sin((float(millis()-starttime)/(60.*60.*24.*1000.))*2.*PI)*30.); // fare opposta ripetto alla temperatura
	h = baseh + rndh + dailyh;

quindi ha un valore base + un valore casuale + un valore con sinusoide con periodo di 24h

 ### Risultati
temperatura
https://test.rmap.cc/graphite/S/FG

umidità
https://test.rmap.cc/graphite/S/FH

i valori istantanei,massimi, minimi e medi paiono coerenti ma molto rumorosi
I valore random aggiunto insrisce rumore, ma la media dei valori random dovrebbe essere zero e mediando su un periodo di 60 secondi il rumore dovrebbe quasi del tutto scomparire

digitecomg, 2023-07-14:
Già modificata l'evoluzione MIN MAX. Riprovata media con simulazione locale con dati in ingresso dal modulo.

pat1, 2023-07-15:
Il simulatore ora genera temperature e umidità con distribuzione gaussiana.
Deve essere effettuato nuovamente il test su un periodo di almeno 24 ore.

pat1, 2023-07-19:
https://test.rmap.cc/graphite/S/FK
effettuato il test.

closed 2023-07-19 by pat1