r-map/rmap #465: stimav4: file info.dat is missed on SDcard

pat1, 2024-01-03:
on SD card the info file is missed.
info.dat:
```
report-gsm
3
10

90
110
```

pat1, 2024-01-03:
sarebbe meglio inserire nel file info.dat la parte costante del topic e ridurre la lunghezza del topic scritta nel file di dati:
info.dat:
```
report-gsm
3
10
1/report/userv4//1162289,4465314/fixed/
51
110
```
2023_08_15.dat:
```
1,0,900/1,-,-,-/B13011
{"v":0,"t":"2023-08-15T00:00:00","a":{"B33199":100}}
2,0,900/1,-,-,-/B13208    
{"v":0,"t":"2023-08-15T00:00:00","a":{"B33199":100}}
2,0,900/1,-,-,-/B13209 
{"v":0,"t":"2023-08-15T00:00:00","a":{"B33199":100}}
0,0,900/1,-,-,-/B14198      
{"v":0,"t":"2023-08-15T00:00:00","a":{"B33199":100}}
254,0,0/103,2000,-,-/B12101
{"v":29239,"t":"2023-08-15T00:00:00","a":{"B33199":100}}
254,0,0/103,2000,-,-/B13003
{"v":84,"t":"2023-08-15T00:00:00","a":{"B33199":100}}
3,0,900/103,2000,-,-/B12101
{"v":29220,"t":"2023-08-15T00:00:00","a":{"B33199":100}}
```

pat1, 2024-01-10:
N.B:
attualmente topic e payload sono scritti uno di seguito all'altro con uno spazio come separatore

info.dat contiene:
* firmware major version
* firmware minor version
* root topic
* topic max len
* payload max len

digitecomg, 2024-03-12:
Sto inserendo info.dat
A questo punto mi viene un dubbio:

Queste le define attive di cui avevamo già discusso dimensioni...

#define MQTT_ROOT_TOPIC_LENGTH (67)
#define MQTT_MAINT_TOPIC_LENGTH (MQTT_ROOT_TOPIC_LENGTH)
#define MQTT_WILL_TOPIC_LENGTH (MQTT_ROOT_TOPIC_LENGTH)
#define MQTT_RPC_TOPIC_LENGTH (MQTT_ROOT_TOPIC_LENGTH)
#define MQTT_SENSOR_TOPIC_LENGTH (38)
#define MQTT_MESSAGE_LENGTH (110)
#define MQTT_WILL_MSG_LENGTH (MQTT_MESSAGE_LENGTH)

char topic[MQTT_ROOT_TOPIC_LENGTH + MQTT_SENSOR_TOPIC_LENGTH];
TOPIC complessivo 67+38 = 105
message[MQTT_MESSAGE_LENGTH]; .... 110

Info dat di esempio ->

nome-stazione (report-gsm)
versione (3)
revisione (10)
topic (1/report/userv4//......../fixed/)
blocco sensor topic (51) ( 1,0,900/1,-,-,-/B13011)        mi sembrava avessimo già detto di stare entro 38
blocco dati (110)

Dubbi:

1) Il 51 è messo ad esempio o è un vincolo minimo di lunghezza calolato perchè altrimenti li devo cambiare (anche se non ci sono stati problemi ad ora in questo senso)
2) il file dati .dat sarà allora formato dalla sola parte sensor_topic (es. 1,0,900/1,-,-,-/B13011) in 51 caratteri (riempiti a NULL), seguito da blocco dati (es {"v":0, ecc.....)) in 110 caratteri riempiti a NULL per ogni singolo record.
Quindi ogni record nel file .dat sarà lungo 51+110 ( o 38+110 )

digitecomg, 2024-03-12:
Comunque inserita la funzionalità, resta da verificare la corretta lunghezza sensor_topic che nel info.dat è impostato a 40 tramite define. Nel caso la cambiamo

closed 2024-03-12 by digitecomg

pat1, 2024-03-12:
Non sono sicuro di aver capito la domanda.
info.dat contiene:

* modello stazione (in stima V3 aveva un significato in V4 mettere il metadato più utile)
* firmware major version
* firmware minor version
* root topic
* topic max len
* payload max len

quindi come esempio info.dat contiene:
* "stimav4 master" oppure "S. Pietro Capofiume"
* 4 
* 2 
* 1/report/userv4//1162289,4465314/fixed/
* 51
* 110

descrizione:
* root topic: la parte invariante per la singola stazione del topic ossia "versione/report/utente/lon,lat/network/"
* topic max len : descrive la lunghezzat (max) del topic variabile dei file di dati
* payload max len: descrive la lunghezza (max) del payload nel file di dati

valori suggeriti:
* topic max len  = MQTT_SENSOR_TOPIC_LENGTH = 38
* payload max len: MQTT_MESSAGE_LENGTH = 110

non c'entra con il file.dat ma MQTT_ROOT_TOPIC_LENGTH  si suggerisce = 50 

i valori da te proposti di MQTT_SENSOR_TOPIC_LENGTH = 40 e e MQTT_ROOT_TOPIC_LENGTH (67) mi paiono comunque compatibili.

digitecomg, 2024-03-13:
Ok, allora va bene come fatto.
Cambio solo il primo metatdato in nome stazione -> S. Pietro Capofiume

digitecomg, 2024-03-13:
Questo il file info.dat creato

paderno
4
5
1/report/userv4//1131085,4445552/fixed/
40
110

All'interno dei files dati Record da 150 caratteri Array[0..149]
Nei primi 40 caratteri
0,0,900/265,1,-,-/B25194 (Riempiti a NULL fino a 40 [0..39])
Nei Caratteri successivi
{"v":79,"t":"2024-03-13T07:00:00","a":{"B33199":100}} (Riempiti a NULL fino a 110 [40..149])