r-map/rmap #468: stimav4: wrong scale for precipitation rate

pat1, 2024-05-22:
I rate di precipitazione paiono moltiplicati per un fattore di scala pari a 10.

infatti 
0.2 Kg/m**2 di precipitazione misurata danno luogo a 120 Kg/m**2/h invece che 12 su base 1'
0.2 Kg/m**2 di precipitazione misurata danno luogo a 24 Kg/m**2/h invece che 2.4 su base 5'

digitecomg, 2024-05-23:
Dalla ISSUE #442

Per la rappresentazione possiamo usare 10^5 e in 14 bit ci stà un valore cumulato mediato di 49 mm in 5 minuti

la risoluzione utile è di 0.1Kg/m^2 in 5 minuti pari a 0,000333333 Kg/m^2/s
Quindi 10^4 dovrebbe essere sufficiente

con 14 bit si rappresentano rate di 16383*10^-4 ~ 1.6Kg/m^2/s pari a una pioggia di 98 Kg/m^2 in un minuto.
Non ci sono molti dati disponibili sul minuto ma piogge di ~ 10 Kg/m in 5 minuti sono frequenti, sul minuto non possono essere altro che valori più elevati e quindi mi pare che l'ordine di grandezza sia corretto.
Probabile che i pluviometri a basculata inizino ad avere seri problemi a basculare 1 volta al secondo ...

Dimmi come meglio fare, se devo dividere per 10 lo inseriamo subito (devo aggiornare i firmware alla 4.2, meglio mettere già questa modifica)

pat1, 2024-05-26:
Malborghetto:
From 22/5/2024, 13:00 Until 22/5/2024, 14:30:00
https://rmap.arpae.it/graphite/S/9
una basculata da 0.1 ogni 3 secondi (mediata in 15')